dmadison/NintendoExtensionCtrl #42: Working then non readable data

tommytippi, 2018-12-16:
I am using the Nunchuck_demo example code
``----- Nunchuk Demi---- Nunchuk Demo -----
The Z button is released
The joystick's Y axis is at 126
The accelerometer's X-ax87) | Buttons: --
----- Nunchuk Demo -----
The Z button is released
The joystick's Y axis is at 126
The accelerometer's X-ax87) | Buttons: --
----- Nunchuk Demo -----
The Z button is released
The joystick's Y axis is at 126
The accelerometer's X-ax+⸮Ȳ⸮P
C⸮⸮󋺑v$⸮x⸮;L⸮i⸮⸮H⸮⸮c⸮0H4⸮R⸮⸮:⸮K6W⸮ (Z3⸮q⸮⸮⸮⸮⸮⸮:⸮h1⸮)⸮S⸮⸮
!Q⸮S⸮⸮B0⸮⸮r,2Ky⸮ZH⸮RP⸮B<M⸮⸮
⸮⸮5
⸮⸮⸮⸮⸮⸮0⸮⸮ (J9⸮I⸮⸮⸮⸮⸮⸮8⸮⸮1⸮⸮S⸮ N⸮⸮⸮I⸮⸮⸮[5⸮R⸮⸮-⸮Iq⸮ӨFp⸮X(⸮⸮4⸮$⸮u⸮F
``
after that i stopped it
i am using arduino uno

dmadison, 2018-12-16:
I can't replicate this. Could you please provide more information?
* What version of the IDE are you using?
* What version of the library are you using?
* Are you using a genuine or knockoff Uno?
* Have you modified the sketch at all?
* How are you connecting to the Nunchuk? If you're using a breakout board, what type is it?
* Are you powering the Nunchuk with 5V or 3.3V power?

Double check that the serial monitor's baud rate is set to 115200. Try using a different USB port / cable, and double-check that your wiring is solidly connected with no shorts.

dmadison, 2018-12-16:
Using a near identical setup (substitute screw terminals for a breakout board), I can't replicate this issue on my end. I ran the demo for 15 minutes and never saw a hiccup on the serial stream.

Your output seems to indicate that the board is having trouble communicating with the PC:

```
----- Nunchuk Demi---- Nunchuk Demo -----
The Z button is released
The joystick's Y axis is at 126
The accelerometer's X-ax87) | Buttons: --
```

Note that last line where the axis value is corrupted, and then it does the "buttons" printout from the end of the `debugPrint` call. It looks like a framing error, where the PC data isn't properly formatted.

Again I would recommend double-checking that your baud rates match (115200), and if you're using a serial monitor other than the Arduino IDE's check that your serial format is correct (8N1). You can also try swapping serial ports or serial cables. If the issue goes away when you disconnect the Nunchuk, then it's likely a wiring problem.

You can also try testing without the library. Upload a sketch that just prints out a long string of text over serial at 115200 baud while keeping the wiring the same (Nunchuk connected and all). Then look for any corrupted data in the stream.

tommytippi, 2018-12-17:
I found the problem, the code was setting the baud rate correctly. however the serial monitor's baud rate was incorrect.

closed 2018-12-17 by tommytippi

dmadison, 2018-12-17:
Glad you got it sorted!