dmadison/NintendoExtensionCtrl #43: Support Question

slashbyte, 2019-01-30:
Hi, really like the library, its written very well. I currently have an NES mini connected to an adafruit feather m0 dev board. The problem I'm having (running the example code) is that the controller is never recognized. any advice you could offer would really be helpful. There is communication over i2c, and if I drop out of the controller check loop the buttons are being read, but it consistently displays the disconnected message on the "term" while displaying valid button data. I was wondering if this might be a clock speed issue? I have ruled out pull up resistors as the culprit. I suspect the library was never meant to run on arm, but it almost dose. Looking forward to hearing back from you.

dmadison, 2019-01-31:
Hey there!

There's no reason the library can't run on ARM or at a faster clock speed. At its lowest level the library uses the `Wire` I2C library to handle communication to / from the controllers, so if the `Wire` implementation for that specific board has issues then the library won't work properly. Although odds are the problem lies somewhere else.

I'm not sure I fully understand what you've written. If the controller is "never recognized" how can you be receiving valid button data? What do you mean "drop out of the controller check loop"? If you run the `NES_DebugPrint` example as-is, does it work properly?

More info:
* Have you tested with another board such as the Uno?
* Are you using a genuine Nintendo brand NES Mini controller or a 3rd party one?
* Are you using pull-up resistors? If so what values?
* Are there any other devices on the I2C bus?
* How are you connecting to the controller? If you're using a breakout board, which one?

slashbyte, 2019-01-31:
Thanks for the prompt response, in the process of answering all of your questions I realized I was running the wrong example... *face palm*. Everything works exceptionally well, I honestly don't know how I over looked that. On a personal note, thanks for publishing the library, its very nice work and thank again for getting back with me, enjoy the rest of your day/night.

dmadison, 2019-01-31:
Haha no worries, happens to the best of us. I'm glad you've got it working, and I'm glad you found the library useful!