dmadison/NintendoExtensionCtrl #45: Who's software am I using?

clplaneguy, 2019-04-10:
I am currently using a library for the Nintendo Wii nunchuck called "Nintendo Extension" in PlatformIO.ini and am looking for documentation.  Is it possible that this library changed its name or registration in the last month?

dmadison, 2019-04-10:
This library's name has never changed. It's always been "Nintendo Extension Ctrl" (see the [blame](../blame/master/library.properties) for the library.properties file).

Is it possible that you misread the title in PlatformIO? It's also possible that PlatformIO isn't parsing the title correctly due to the whitespace. Looking at the PlatformIO libraries search [my library is the only library I found searching "Nintendo Extension"](https://platformio.org/lib/search?query=nintendo%20extension).

I've never used PlatformIO myself, but I'd recommend looking at the source files you're using and see what's in them. The source files for this library all have [an LGPL header with my name and a link to the repository](../blob/master/src/NintendoExtensionCtrl.h#L1-L21).

clplaneguy, 2019-04-10:
The link gives me an error

*Redirect Notice*

 The previous page is sending you to an invalid url (http://
../blame/master/library.properties).

 If you do not want to visit that page, you can return to the previous page
<https://www.google.com/url?q=http://../blame/master/library.properties&source=gmail&ust=1555003826682000&usg=AFQjCNHecP99EUC5KIocTFLT9PmAkelQ9w>
.

The blame on library.properties shows that the name has been consistent for
the last year.  I installed your library less than a year ago.

I can see that the name has been "Nintendo Extension Ctrl" since version
0.0.1!

I am talking to PlatformIO in this thread.  You may be interested.

https://community.platformio.org/t/i-need-documentation-for-a-public-library/7301

I am discussing that I have a call to "Nintendo Extension"  that works.
But no one seems to know where it is going or how.

dmadison, 2019-04-10:
I'm not sure what to tell you, the links work for me. They're relative, so you might have to click them on GitHub rather than through your email.

Looking at the PlatformIO thread, my guess is that you're using this library but the name got truncated somehow.

clplaneguy, 2019-04-10:
When I check the compile log I see

```
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 16 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <MakeBlockDrive> 3.26 #b7b9738
|   |-- <Wire> 1.0
|-- <PID2> #dd51ba7
|-- <SoftwareSerial> 1.0
|-- <Nintendo Extension Ctrl> 0.7.2
|   |-- <Wire> 1.0
|-- <Wire> 1.0
Checking size .pioenvs\megaatmega2560\firmware.elf
```

**Thank You for a useful library.**

Just in case.
My project is hard stuck.  Perhaps you have some expertise with PID.
My Arduino thread is at
`https://forum.arduino.cc/index.php?topic=608412.30`

dmadison, 2019-04-10:
Great, I'm glad you solved your issue and that the library is proving useful to you. I'm going to close this issue as your problem appears solved.

Unfortunately I don't have the time to help out with other projects. Good luck!

closed 2019-04-10 by dmadison