dmadison/NintendoExtensionCtrl #46: 8BitDo Retro Receiver gives garbage data

nullstalgia, 2019-12-21:
Hi!

I was hoping I could use your code with an 8bitdo Retro Receiver for (S)NES Classics, but I don't get much anything useful from it.

(These were tested with/without I2C pullups with a value of 4.7KOhm)

Output from IdentifyController with Adapter

```
Classic Controller connected!
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
```

With Official NES Classic controller

```
Classic Controller connected!
ID: 0x01 0x00 0xA4 0x20 0x01 0x01
```

With Official Wii Classic controller

```
Classic Controller connected!
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
```

With Official SNES Classic controller

```
Classic Controller connected!
ID: 0x01 0x00 0xA4 0x20 0x01 0x01
```

With 3rd party Wii Classic Pro Controller

```
Classic Controller connected!
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
```

Output from Classic Controller Debug Print

With Official SNES

```
Classic ____ | ___ | ____ L:(31, 31) R:(15, 15) | LT: 0_ RT: 0_ Z:__
```

With 8BitDo

```
Classic <^v> | -H+ | ABXY L:( 4,  6) R:(21,  6) | LT: 4X RT: 6X Z:LR

```

Output from printDebugRaw

8BitDo

```
Raw[6]: 0x84 0x86 0x86 0x86 0x00 0x00
```

8BitDo with 8-sized request data

```
Raw[8]: 0x84 0x86 0x86 0x86 0x00 0x00 0xFF 0xFF
```

SNES
```
Raw[6]: 0x5F 0xDF 0x8F 0x00 0xFF 0xFF
```

dmadison, 2019-12-21:
Hey there! Thanks for the detailed report.

When requesting 8 bytes from the 8BitDo, do the last two bytes change when you press the buttons?

nullstalgia, 2019-12-22:
Yes, actually! It seems to be mapped like a normal Pro Con, including zr and zl being seperate

~~But for some reason, if I use a compatible controller with analog sticks., the Left seems to be mapped to the DPad while the Right does nothing.~~

~~It's possible thats just how it acts on purpose, but when connected directly to a PC with the XInput mode of the 8bitdo, it has full dual stick functionality.~~

Connecting it to an official Wiimote shows the same stuff, so I guess this is just how it's meant to be used.

dmadison, 2019-12-22:
This looks like a continuation of a known problem (see #33). For whatever reason, the 3rd party NES controllers do a poor job of emulating the genuine article and in unencrypted mode they don't behave properly. They send junk data for the first six bytes and put the button data they *do* have into bytes 7 and 8.

The library is set up to handle this (#37), but all of the 3rd party NES controllers I've seen so far display a different data signature than that 8BitDo receiver so the library wouldn't have tried to fix it.

I've pushed some changes to a new branch that will hopefully work for you. Checkout the [nes-thirdparty](../tree/nes-thirdparty) branch and try one of the NES examples. Let me know if it works for you and I'll merge those commits to master.

nullstalgia, 2019-12-22:
```cpp
#include <NintendoExtensionCtrl.h>

ClassicController nes;

void setup() {
	Serial.begin(115200);
	nes.begin();

	while (!nes.connect()) {
		Serial.println("Classic Controller not detected!");
		delay(1000);
	}

	if (nes.isNESThirdParty()) {  // Uh oh, looks like your controller isn't genuine?
		nes.setRequestSize(8);  // Requires 8 or more bytes for third party controllers
	}
}

void loop() {
	boolean success = nes.update();  // Get new data from the controller

	if (success == true) {  // We've got data!
		nes.fixNESThirdPartyData();  // If third party controller, fix the data!
		nes.printDebug();  // Print all of the values!
	}
	else {  // Data is bad :(
		Serial.println("Controller Disconnected!");
		nes.reconnect();
	}
}
```

Works perfectly! Only request with this specific issue would be to maybe not have `fixNESThirdPartyData` and `fixThirdPartyData` be separate if it doesn't exclusively apply to NES controllers anymore, but that's just me.

---

Additionally, would you mind if I opened another issue with another quick feature request? I've seen that you can choose the Wire "port" with `Class controller(Wire1);` 

I'd love to have that same functionality, but with the Serial port. So by default it'd be Serial and Wire, but if I just need to change the Serial port, it could be `controller(Wire, Serial1)` so that way you don't have too many different overloads. :)

dmadison, 2019-12-22:
The fix is exclusive to S/NES controllers, which is why it's part of the function name. Even though you're using the `ClassicController` class, only the buttons are functional from the 'fix'. The analog inputs are overwritten to their defaults, even if the physical controls are present.

Create another issue for the feature request and we can talk about it there 😊.

nullstalgia, 2019-12-22:
Hey, as long as it works, all seems good to me! Thanks so much for your help and promptness!

Feel free to close this when you merge it/whenever you want.

dmadison, 2019-12-23:
Briefly following up: you mentioned that you're trying to connect the Classic Controller to the Switch. Are you using any of the analog inputs of the 8BitDo receiver? I'm wondering if it might be better to try and create a separate controller class if you can figure out the mapping.

nullstalgia, 2019-12-24:
I'd like to use the Analog Inputs, but it seems that it doesn't output them at all when using I2C, at least not with the hardware I've been trying (Official Wiimotes and Arduinos).

nullstalgia, 2019-12-24:
From the manual: 
[image]

It seems that they intentionally map the Left Stick to DPad, and the Right to nothing.

dmadison, 2019-12-24:
Ah okay, just wanted to make sure we aren't overlooking anything. Thanks for the info.

closed 2020-01-05 by dmadison