dmadison/NintendoExtensionCtrl #47: Specifying Serial port for .print functions during declaration?

nullstalgia, 2019-12-22:
So, I've been working on a project with your library that basically connects a Classic Controller to a Nintendo Switch via progmem's excellent work [here](https://github.com/progmem/Switch-Fightstick) and the Arduino version by fluffymadness [here](https://github.com/fluffymadness/ATMega32U4-Switch-Fightstick/).

The thing is, it requires using LUFA which removes the Serial port in favor of Serial1 and Serial2. 

Now, I can change it in the `NXC_Comms.h` header, but when I want to use the library/test something with normal Serial and I forget that I changed the port, I get no output until I revert the change.

--- 

Copy from issue #46

I've seen that you can choose the Wire "port" with Class controller(Wire1);

I'd love to have that same functionality, but with the Serial port. So by default it'd be Serial and Wire, but if I just need to change the Serial port, it could be controller(Wire, Serial1) so that way you don't have too many different overloads. :)

dmadison, 2019-12-23:
I'm hesitant to add that because it means needlessly storing the reference to the Print object. The only functions that use the Print object are debugging ones, and for the vast majority of people they're only going to use the Serial (USB) interface.

For your use in particular, you could create a preprocessor statement to catch if LUFA is being used in the Comms header. E.g.:

```cpp
#ifdef(LUFA)
#define NXC_SERIAL_DEFAULT Serial1
#else 
#define NXC_SERIAL_DEFAULT Serial
#endif
```
I only have a little bit of experience with LUFA in the context of the Arduino build environment, but from what I recall it does set up some global defines that you could use.

nullstalgia, 2019-12-24:
I tried something like that and I'm very confused...

In NXC_Comms.h
[image]

With LUFA
[image]

Without LUFA
[image]

---

In NXC_Comms.h
[image]

When using LUFA
[image]

dmadison, 2019-12-24:
Not sure what to tell you there.

dmadison, 2020-01-05:
Were you able to solve the issue?

nullstalgia, 2020-01-09:
Ah, sorry! I had to do something a lil weird but it ended up working for me.

I added a flag to the compiler script for LUFA

(for anyone who's curious...:

`/usr/share/arduino/hardware/lufaduino/avr/platform.txt`

And in platform.txt, under "AVR Compile Platform", I edited the "compile c++ files" command to 

`## Compile c++ files
recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mmcu={build.mcu} -DF_CPU={build.f_cpu} -DLUFAON=1 -DARDUINO={runtime.ide.version} -DARDUINO_{build.board} -DARDUINO_ARCH_{build.arch} {compiler.cpp.extra_flags} {build.extra_flags} {includes} "{source_file}" -o "{object_file}"`

The only change from the normal command is `-DLUFAON=1` in the middle. 

Then in NXC_Comms.h I added

```cpp
#ifdef LUFAON
#define NXC_SERIAL_DEFAULT Serial1
#else
#define NXC_SERIAL_DEFAULT Serial
#endif
```

---

`lufaduino` is a copy of `archlinux-arduino` with LUFA patch applied, then copied to its own folder in `/usr/share/arduino/hardware`  . Then I unpatched the normal `archlinux-arduino` so they are separate.

Then I edited boards.txt to remove everything but the Micro, and platform.txt to change the name of the platform

[image]

)

dmadison, 2020-01-09:
Fantastic! That looks like a good solution, and I'm glad you were able to sort it out.

---

Before I close this: I don't think either of us mentioned it in the thread, but it is already possible to override the Serial object for debug printing. You just need to pass the new `Print` object to the debug function:

```cpp
nunchuk.printDebug(Serial1);  // prints debug output to Serial1
```

This gets unwieldy for multiple controller objects and multiple function calls, which is why @nullstalgia created the issue.

closed 2020-01-09 by dmadison