dmadison/NintendoExtensionCtrl #50: Genuine NES Classic Controller hangs at connection time. Any ideas?

epiblitikos, 2020-01-14:
Hi, I'm having trouble getting a genuine NES classic controller to connect on a genuine Arduino UNO board. I annotated the related print-debug sketch as follows.

```cpp
void setup() {
  Serial.begin(115200);
  Serial.println("beginning nes");
  nes.begin();
  Serial.println("nes has begun");

  Serial.println("connecting controller");
  while (!nes.connect()) {
    Serial.println("Classic Controller not detected!");
    delay(1000);
  }
  Serial.println("controller has connected");

  Serial.println("detecting 3rd party");
  if (nes.isThirdParty()) {  // Uh oh, looks like your controller isn't genuine?
    nes.setRequestSize(8);  // Requires 8 or more bytes for third party controllers
  }
  Serial.println("detection complete");
}
```

This produces the following output when run with the controller connected.

```
15:55:39.474 -> beginning nes
15:55:39.474 -> nes has begun
15:55:39.474 -> connecting controller
```

This leads me to believe there's something hanging in the connection step.

Given everyone else's success with the lib, I'm sure something in my setup is incorrect, but as far as I have found, I'm wired up correctly. Basically I bought an extender, then snipped and stripped that to make a female port on the breadboard. I then plug the controller into that. Here's a shot of Fritzing for reference.

[image]

Any help you can provide resolving this is greatly appreciated! Cheers!

epiblitikos, 2020-01-14:
Oh, and for what it's worth, I get the same result when I disconnect pin 3 on the connector.

dmadison, 2020-01-14:
Hey there! Sorry that you're having issues, and thanks for the detailed report.

It's likely that you have an issue with your wiring. Take a look [at this image](https://wiibrew.org/wiki/File:Wii_Connector.jpg) and see if your connections match up. Do you have a multimeter? If so you should check that you have continuity on each pin from the downstream connector to the wire that plugs into the Arduino.

Hanging on the `connect` function is a sign that the I2C function calls aren't returning, which will happen if something is holding the bus lines low.

You also do not need the connection to pin 3. That's used for checking whether a controller is connected, and does not need to be tied to anything.

epiblitikos, 2020-01-14:
Thanks for the help and the fast reply! I'll double check my wiring with a meter--and thanks for the confirmation about pin 3.

epiblitikos, 2020-01-15:
Yeah, turns out I had a couple of the pins to wires mixed up. Thanks again!

closed 2020-01-15 by epiblitikos