dmadison/NintendoExtensionCtrl #51: Genuine Classic Controller does not Initialize/Connect properly with your XInput AVR Library

nullstalgia, 2020-01-26:
Hey, me again :P

(Note, this has only been tested with a custom ATMega32u4-based board)

So I've been using your XInput library (which I didn't notice was yours, even after I used the Classic Controller > XInput example), and I noticed an odd bug.

When I have a normal, genuine Classic Controller plugged in when the AVR resets, the 

```cpp
while (!classic.connect()) {
    digitalWrite(6, !digitalRead(6));
    delay(1000);
  }
```

Will blink the LED over and over again until I do one of the following:

- Replug the controller
- Reset the chip with a SNES/NES/Knockoff controller plugged in
- Do a full power cycle with the Classic controller plugged in

This seems similar to #50, but it seems to work fine with the NES Classic Controller no matter what. (I have 4.7k Pullup resistors on the board)

Unfortunately, it seems to work fine occasionally.

But the ways I've been able to make it mess up seem to be this:

- Have a stick not be centered on reset
- Upload a new sketch
- Sneeze on Tuesday when the moon is waxing

Any help would be appreciated :)

dmadison, 2020-01-26:
Hey there again!

I've noticed this behavior as well when I was first building the Classic Controller support. As I remember it, when the controller has already been initialized with the unencrypted register settings it no longer reports back the same "identity" string. The controller-specific classes that don't use a shared data instance will report back a failure to connect if the identity string doesn't match what it expects, so it will never exit that 'connect' loop.

I believe the identity issue was unique to the Classic Controller, but all of the extension controllers have similar issues if you reset the board in the middle of an I2C write. It messes up the bus processing and the controllers refuse to respond to subsequent commands until they're powered down and reconnected.

This _should_ only be an issue during development though, because the board should normally never have to reset while the controller is connected and powered.

nullstalgia, 2020-01-26:
Hmm, ok. I'll keep an eye on this, and hopefully it won't be an issue when I start sending out test units :)

On the topic of the Classic Controller, I noticed you have the "bounds" of the sticks set with

```cpp
XInput.setRange(JOY_LEFT, 8, 56);
XInput.setRange(JOY_RIGHT, 4, 28);
```

And I tried using those Min/Max values in a similar sketch, but I've seen the values go past both bounds in both directions! I remember for a while my controller actually wouldn't go past 56/28, but now it does seem to have the full 0-63 and 0-31 resolution.

Is there a more grounded reasoning behind those bounds or should I stick to the 0-X?

dmadison, 2020-01-26:
The XInput range functions will take inputs outside of the set range and clip them to the max values. The joystick range values are set slightly inwards because for most games it's more important to be able to hit the min/max X/Y levels than it is to have that little bit of extra precision. Plus not every controller will reach the min/max of the available range - the Nunchuks in particular are notorious for this.

> hopefully it won't be an issue when I start sending out test units :)

I don't know what you're building, but bear in mind that the VID/PID identifiers used with the XInput library are strictly for non-commercial projects! If you're building a commercial product you should contact Microsoft for licensing.

nullstalgia, 2020-01-26:
That makes sense, yeah. My issue is that when using map() with the in_min set to `4` and it hits `0`, it flings the output to the exact opposite side. I might have to do some work on the input then.

And ah, shoot. Maybe it could just be an alternate firmware that is supplied free of charge. I'm working on an adapter that converts a SNES/NES Classic controller and makes it work natively with the Nintendo Switch (via [progmem (and team?)'s excellent reverse engineering](https://github.com/progmem/Switch-Fightstick) and PC via your XInput library. However, since one uses LUFA and the other is... outside of my comprehension, I had to find a way to put the two firmwares on one AVR. 

I found [mihaigalos's miniboot](https://github.com/mihaigalos/miniboot) (which is a custom bootloader that can reflash an AVR with code from an external I2C EEPROM) and planned on having 1 or 2 EEPROMs to store that code.

[image]
[image]

Prototypes have worked well so far, and even had a first PCB made. 

That non-commercial portion could be a hitch, though.

dmadison, 2020-01-26:
You just need to check if the value is out of range before you pass it to `map` for rescaling. Here's the XInput rescale function, for reference:

```cpp
int32_t XInputController::rescaleInput(int32_t val, Range in, Range out) {
	if (val <= in.min) return out.min;  // Out of range -
	if (val >= in.max) return out.max;  // Out of range +
	if (in.min == out.min && in.max == out.max) return val;  // Ranges identical
	return map(val, in.min, in.max, out.min, out.max);
}
```

You could always use DirectInput as an alternative, although if you're planning on selling the board you'd still need to acquire your own USB VID/PID. Although I'm sure that's cheaper than licensing an XInput product.

---

Regardless, I think we've veered off topic from the original issue. In a nutshell, the extension controllers were designed to connect to the Wiimote, where the host and extension controller were operating from the same power supply and could only reset together. It was never designed to talk to a host that could spontaneously reset in the middle of communication without also resetting the state of the connected controller - so you're bound to run into issues. 

Every time I've run into this problem it was caused by having the microcontroller reset without also resetting the controller. Disconnecting and reconnecting the controller solves the issue immediately.

nullstalgia, 2020-01-26:
Ohh, that's a pretty slick implementation, thanks for the reference!

And fair enough with the DirectInput idea, that actually whooshed past me. And I'd love to get a VID of my own, but a look at http://pid.codes/ would probably be better for me, as I don't think small Tindie sales can make up multiple thousand dollars :P

---

And that is a fair point. It did seem that the issue was only appearing with the XInput library, but it's possible that it was just confirmation bias. I'll have to look into it more.

In the meantime, feel free to close the issue if you don't have any further questions/suggestions. Additionally, I'll trim down the previous reply to make it less "self-promote"-y.

dmadison, 2020-01-26:
Yeah, that definitely sounds like a better idea to me. DirectInput has fallen out of vogue but it is still the best _open_ HID standard we have.

Definitely look into it some more and let me know if you find any problems or incompatibility issues with this and the XInput library. Anything is possible, but from my own prior testing I think these problems are unrelated to the XInput library.

Closing for now. Feel free to re-open if you find more information.

closed 2020-01-26 by dmadison