dmadison/NintendoExtensionCtrl #52: Not an issue - but...

phaleas, 2020-02-14:
I'm using your excellent library with an ESP32 - after using it with Arduinos before. It seems, that you changed some files within the current version 0.7.4. These changes effect the users of ESP platform heavily, and many of them would not expect changes within one version. Please build a 0.7.5, so that ESP users get the latest changes (which are working like charme).

dmadison, 2020-02-14:
I'm sorry, I don't follow. The latest commit updates the repository to version 0.7.4 and is a minor change to the DJ Hero's "crossfade" function. There have been no commits/changes past that point on any branch.

What changes are you seeing, and how do they disproportionately affect the ESP platform?

phaleas, 2020-02-14:
I'm so sorry. I had multiple Github pages open in tabs (different Arduino libraries), and posted an issue of another library (Encoder) in your issue tracker.

I've had no issues at all with your library.

dmadison, 2020-02-14:
No worries, happens to the best of us. I'm glad the library is working well for you!

closed 2020-02-14 by dmadison