dmadison/NintendoExtensionCtrl #53: Classic Controller not detected on Wemos D1 mini boards

nordblick2, 2020-02-15:
Hey Dave,
some days ago I found your lib and it works like a charm with my cheap chinese classic controller pad. For my first test I used an Arduino Micro. 

For the main project (a self coded tetris clone) I want to use an ESP8266 (Wemos D1 mini) especially for its wifi support. Unfortunately same example code (that worked on the Micro) won't work on my D1 board. I think, its not an core i2c or wiring problem, because an BME280 sensor I put on the same i2c line works without problem. I even tried to ad the usual 4k7 pull-ups - without any change. The .connect() always returns false (without any delay), and I have no idea how to fix it. 

For debugging I also used the usual i2c-scan sketch that that found an device on 0x52 - obviously the controller. anyway, communication seems not possible on the D1. Do you have any idea why this issue appears on my D1 board? Is that a similar problem as in issues #33?

Cheers.
Olli

nordblick2, 2020-02-15:
And here some additional information from the logic analyser (same code, different board):

Arduino | Wemos D1
-- | --
Setup Write to [0xA4] + ACK | Setup Write to [0xA4] + ACK
0xF0 + ACK | 0xF0 + ACK
0x55 + ACK | 0x55 + ACK
Setup Write to [0xA4] + ACK | Setup Write to [0xA4] + ACK
0xFB + ACK | 0xFB + ACK
0x00 + ACK | 0x00 + ACK
Setup Write to [0xA4] + ACK | Setup Write to [0xA4] + ACK
0xFA + ACK | 0xFA + ACK
Setup Read to [0xA5] + ACK | Setup Read to [0xA5] + ACK
0x00 + ACK | 0x00 + ACK
0x00 + ACK | Setup Write to [0xFE] + NAK
0xA4 + ACK | 0xFE + NAK
0x20 + ACK | 0xFE + NAK
0x01 + ACK | 0xFE + NAK
0x01 + NAK | 0xFF + Missing ACK/NAK
Setup Write to [0xA4] + ACK |  
0x00 + ACK |  
Setup Read to [0xA5] + ACK |  
0x20 + ACK |  
0x20 + ACK |  
0x10 + ACK |  
0x00 + ACK |  
0xFF + ACK |  
0xFF + NAK |

nordblick2, 2020-02-15:
OK, I guess, I resolved it: Wemos D1 (all ESPs?) seems just to be on speed 😜. ESPs/D1's default i2s speed is 400k. Everything's working with 

```
Wire.begin();
Wire.setClock(10000L);
controller.begin();
// ...
```
Will do further test tomorrow and (probably) close that issue.

dmadison, 2020-02-16:
I don't have a D1, but I just tested with a NodeMCU ESP8266 and didn't find any issues.

My guess is that it's a pull-up problem, particularly if reducing the speed helps. All of the Nintendo extension controllers use pull-ups on the controller itself rather than the host because the wire length (bus capacitance) varies wildly depending on the controller. Chinese controllers are notorious for foregoing these pull-ups. I'd recommend using a multimeter to check the pull-ups on your own Classic Controller.

You mentioned trying some 4.7k pull-ups so you may have the opposite problem with the pull-ups being too *strong*, particularly if you have another device on the same bus with its own pull-ups.

The AVR Arduinos also enable their internal pull-ups for the I2C bus by default (~30K) while other microcontrollers don't. Because the primary pull-ups are on the controller and not the master, some microcontrollers like the Teensy will hang waiting for I2C response if the controller is disconnected since it leaves the bus floating. A weak pull-up fixes this - I've had good results with ~47K.

Lastly, the genuine Classic Controllers don't like being re-initialized. If you reset the microcontroller without powering down the controller, the `connect` function may return `false` because the controller's identity string no longer matches the Classic Controller. Power cycle the controller to fix the issue. I don't think that's your problem, but that issue may interfere with your testing.

Do you have access to an oscilloscope? I would look at the signal directly if you can.

dmadison, 2020-02-29:
It's been two weeks without an update, I'm assuming this is solved. Feel free to re-open if you still have issues.

closed 2020-02-29 by dmadison