dmadison/NintendoExtensionCtrl #63: Change success condition for getHighRes/setHighRes

nullstalgia, 2020-05-26:
First off, great work!

Secondly, setting hi-res on a crappy third party didn't work, but it has an easy solution:

Check to see if the ID of the controller changed to reflect the new mode

On a controller that supports HiRes:

[image]

On a controller that doesn't:

[image]

Tested with:

Official Classic Controller

Official SNES Mini Controller

8BitDo Retro Reciever

Crappy Classic Pro

dmadison, 2020-05-28:
I swear, these crappy Chinese controllers will be the death of me.

Unfortunately that isn't an easy solution because it breaks the connection flow for all Classic Controllers. The assumption is that if you write a byte to a register and the controller acknowledges it, the setting is saved and the library adjusts accordingly. If the data is accepted (ACK) but the controller *does not update accordingly* the communication flow breaks down.

The library doesn't have a method by which to say "the controller is present but this mode isn't available" - the `connect` function is a simple pass / fail for the controller you're using. If you make the `highRes` mode conditional on whether the controller ~~is not a piece of junk~~ supports the given communication mode, controllers that only support one or the other will return a fail condition to the connection function even if they're communicating data.

One option would be to have the library automatically juggle this - match its internal value to the mapping type supported by the controller. The problem there is that you're silently changing the output ranges on the user. They would have to check the value of the `highRes` variable themselves on every connection and change their output mappings accordingly, even if they're only trying to operate in one mode.

Another option would be to roll back the existing implementation and rebuild this from the ground up, marking a strict division between the "normal" and "high res" modes by using separate classes for each. As above, controllers would fail if their registers didn't match the expected mode on connection but this would instead be an advantage, because you only need to worry about the ideal condition. On the other hand, this would require two objects if you wanted to support all 3 versions of the controller (genuine, KO Wii, KO Mini). It also introduces a different issue, because you can't have separate type (enum) definitions for the two versions since the "real" controllers always ID in standard mode, which breaks the `MultipleTypes` example and iterating through the `ExtensionController` arrays.

You could also store the states separately - one for what mode the controller is in and one for what mode the user *wants* it to be in. Then scale the values one way or the other to match the output levels. Which just seems Rube Goldberg-esque to me...

nullstalgia, 2020-05-28:
Well, the way I was thinking, is to write to the I2C register of the controller, as normal, then read it back to see if it changed. And if it didn't change like we wanted it to, we return a `false` and move on.

Also, wouldn't it be possible to maybe give the ClassicController class its own 'specificInit' and check to see if the controller supports it via that method? This wouldn't be needed in the MiniController class since it's always going to be needed.

Additionally, you could have the user (or class?) perform a sanity check on the values being given. If I'm understanding the data layout correctly, the DPad values are stored bytes 7 and 8 (the last 2), and normally, a controller not in Hi Res mode would probably output the same stuff constantly. (Not unlike how you did the check before in NESThirdParty.) If those are all pressed down (which is impossible given the physical designs), then they could set a hard "no" to hiResMode until the controller is disconnected.

---

The reason I bring this up, is because of that project I told you about/sent you on Twitter. I want it to be compatible with both crappy and original controllers, and it uses the ClassicController class since it's compatible with both Mini pads and normal Classic pads, but setting it to HiRes just for the Mini pads seems like a waste (and might be difficult to differentiate) if it can give extra resolution to normal Classic pads.

It feels like this is the final hurdle we have to overcome :)

dmadison, 2020-05-29:
>Well, the way I was thinking, is to write to the I2C register of the controller, as normal, then read it back to see if it changed. And if it didn't change like we wanted it to, we return a false and move on.

The problem is *where* you return that 'false' to. If it's part of `specificInit` and done during the connection check then the connection fails even if the controller can communicate and work with the alternate mode. Your "Crappy Classic Pro" that started this issue for example would never be able to connect so long as you're forcing the high res mode. Conversely, the knockoff NES Minis would never connect to the standard Classic Controller class.

>Additionally, you could have the user (or class?) perform a sanity check on the values being given.

But to what end? If they want to the user can already perform a sanity check on the values given just by AND-ing the DPAD functions. Doing some sort of a data check on update has the same problem: if it's automatic, it would change the data layout and not inform the user. And for programs that use multiple types you would have to call that class-specific update function rather than the generic port one. Not to mention that it would require a virtual function, which is yet again more overhead...

You're also assuming here that the user won't be using a modded or third party controller like a fight stick that represents itself as a Classic Controller but doesn't have the pivot in the center of the DPAD.

---

To be honest my gut reaction is to say "screw the crappy controllers" and restructure the Classic Controller class into two halves for "standard" and "high res" modes (see the commit history on that pull). Then force the data mode on connection and let the chips fall where they may.

nullstalgia, 2020-05-29:
> and done **during the connection check** then the connection fails even

Oof, this is a part that I missed. I haven't fully tested this update, and I wasn't aware that it was packed into `connect()`. I thought it was a separate function, like:

```cpp

controller.begin();
controller.connect();
controller.specificInit();
```

That's my mistake, and now I see the issue.

My gut reaction would be to return a numeric/enum value with connect(), but that would break code that uses `connect() == true` so that is immediately a no-go. 

There could also be a `supportsHiRes` public bool that is populated on `connect` where it runs that test I stated above?

---

Another thing that came to mind, and feel free to shoot it down, is maybe using floats instead of the raw ranges.

I'm not sure if they can go negative, so it could be

```
0 far left,

1 center,

2 far right
```

So that way its a dynamically scaling range that wouldn't care if the class suddenly went into HiRes mode without telling the user.

I dunno, just something to think about.

nullstalgia, 2020-05-29:
Since I have a Bus Pirate and a couple Classic Minis, I might hook it up as a sniffer and see if I can watch what it does with a knockoff controller that can't use the HiRes mode.

I'll report back if I do, maybe tonight.

dmadison, 2020-05-29:
>I haven't fully tested this update, and I wasn't aware that it was packed into connect()

It's both - it can be called on its own (again with the `MultipleTypes` example) but it's normally called at the end of a successful connection to do any controller-specific initialization that's required. This lets us do some behind the scenes magic (like with the Drawsome tablet's registry writes) without any extra input from the user.

>There could also be a supportsHiRes public bool that is populated on connect where it runs that test I stated above?

Wouldn't that give you the same result though? If you're trying to initialize the controller as "high res" and it doesn't support it, you're back to either having the user intervene or changing the data ranges automatically.

>Another thing that came to mind, and feel free to shoot it down, is maybe using floats instead of the raw ranges... So that way its a dynamically scaling range that wouldn't care if the class suddenly went into HiRes mode without telling the user.

I had this thought as well, although using integers: if you're in the low resolution mode, bit shift or rescale the analog data to fit a full byte. Then the user-facing range doesn't change even though the background mode does.

I'm not fond of the idea because you're manipulating the output data, and I'd like to keep the library just as a pass-through. For example there's no automatic min/max range rescaling for the other controllers either but it's easy enough to add on top of the library if the user needs it.

But in spite of the added complexity and overhead, the flexibility is appealing...

>Since I have a Bus Pirate and a couple Classic Minis, I might hook it up as a sniffer and see if I can watch what it does with a knockoff controller that can't use the HiRes mode.

What are you looking for? From what I've seen on my logic analyzer the I²C command is written and accepted, the controller just doesn't change its internal register.

Your initial suggestion to change the success condition is a good one. It's the effect on the rest of the communication flow that concerns me.

dmadison, 2020-05-29:
>What are you looking for? From what I've seen on my logic analyzer the I²C command is written and accepted, the controller just doesn't change its internal register.

Apparently I was misremembering this. Just tested with my NES Mini knockoff and writing to that register saves the value but still only supports "high resolution" mode. Damn.

I think this may bust... too many different controller knockoffs out there with varying behavior.

The only solution I can think of is to try to set the mode value, check it against the register, perform a control update, and then run that subsequent data through a sanitization function to check that it matches what you're expecting. Then if it doesn't, re-run the entire process the other way around. Which is messy to say the least, and again we're back to some sort of a `checkIfCrappyController` function.

nullstalgia, 2020-05-30:
Yeah, I had no luck getting any data from it. I tested a lot of controllers, and when I switched to the crappy one, it just didn't do anything. So that's fun :/

Frankly, it's not looking good for those.

It's your library, so it's up to you. :)

I'll help/test however I can!

dmadison, 2020-06-05:
Fingers crossed that I've got it sorted. Give the `classic_flex` branch a try and let me know if it works for you.

All of the 'standard' analog data maps have been remapped into the high res range (0-255 each) and the data mode is read on connection (via `specificInit`) by comparing bytes 7 and and 8 in the control data array to `0x00`. The logic is that since those bytes are only used in 'high res' mode, the standard bytes should always be `0x00`. This temporary read is done twice and cross-checked to try and mitigate the effects of noise.

See this commit for details: 91d870ec9f895add0fe70bda4f8769a7b212464d

The advantage is that the user-facing class:

* Should work with *all* Classic-type controllers (genuine, Classic K/O, NES K/O, SNES K/O, custom 3rd party)
* Maintains the same data ranges regardless of mode
* Requires no extra input or attention from the user to adjust to different controllers

The disadvantage of course is the longer and more intensive `connect` function, plus a bit of extra overhead for the analog input bitshifting. But I think that's a small price to pay if it truly does support all of the variations out of the box.

This works for the three controllers I have on hand (genuine Classic Controller, NES knockoff, SNES knockofff) but I don't have a knockoff version of the Classic Controller to test with. Please test with your controllers and let me know if you have any issues.

nullstalgia, 2020-06-07:
[image]

Tested controllers and cat.

From left to right:

8BitDo SN30 (via **8BitDo Retro Reciever**) - Works †

Hyperkin SNES/SFC Controller to Mini Classic - Works ††

K/O Classic Pro - Works

Genuine Classic - Works

Genuine SNES Classic - Works

~~Cat - Not compatible.~~

† - L and R sticks aren't completely centered, but is likely due to the adapter itself. Having the user do an automatic "deadzone" based on the initial position is likely best. (Which is what I do in my adapter anyway)

†† - DPad acts like L stick, but that's just the adapter. I tried to force it to nonHighRes, but as I haven't taken more than a glace into the code yet, it's not a fan of that. :sweat_smile: 

```
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:255_ Z:LR | (HR)
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:255_ Z:LR | (HR)
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:255_ Z:LR | (HR)
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:255_ Z:LR | (HR)
``` (Swapping each update cycle, while holding buttons. Seemingly nothing changes)

nullstalgia, 2020-06-07:
So if I had to make one suggestion, which is entirely optional given that these seem to work fine, but someone may want this:

Is to allow a *forced* mode for High/Low res. So that no matter what the controller outputs, it will be in and *stay* in that mode. (For example, with the current branch, I can't tell if the Hyperkin would react the same in non High Res.) I could take a crack at it with a `protected` bool that the `get` `set` and `check` hiRes functions would peek at first, but if you want to do it yourself in a way that would mesh better with your style, that works too. Or, just have `setHighRes` have a `force` argument that's false by default. 

Lemme know what you think :)

dmadison, 2020-06-07:
Thanks for testing! I appreciate the info.

>8BitDo SN30: L and R sticks aren't completely centered

How far off are they? If the adapter isn't reporting the sticks as centered there's not much we can do, but hopefully it's only a few degrees off and should be caught by most software deadzones.

>I tried to force it to nonHighRes, but as I haven't taken more than a glace into the code yet, it's not a fan of that.

What exactly is happening there? Is that "swapping" output just the standard `debugPrint` from the example? Have you had a chance to look at the I2C data?

>[One suggestion] is to allow a forced mode for High/Low res. So that no matter what the controller outputs, it will be in and stay in that mode.

At the moment the mode switching function is called only on 'connect' - the subsequent data from updates is not parsed and the internal `bool` to change between modes is never changed unless the user explicitly calls `setHighRes`.

The only 'forcing' that could be done is telling the class to stick to the map you're trying to set regardless of whether the controller is outputting the data that matches it - which frankly doesn't make any sense to me. At that point the issue is with the `check` function not detecting the right data mode, which from your description doesn't seem to be the case?

>~~Cat - Not compatible.~~

I'll get to work on that next :kissing_heart:

nullstalgia, 2020-06-07:
> How far off?

`Classic ____ | ___ | ____ L:(132, 134) R:(134, 134) | LT:  0_ RT:  0_ Z:__ | (HR)` 

Outputs the same whether or not a controller is connected to the adapter or not.

> Is that "swapping" output just the standard debugPrint from the example?

~~Essentially. It's the same minus this middle snippet.~~

```cpp
bool hr = false;

void loop() {
	boolean success = classic.update();  // Get new data from the controller

	if (success == true) {  // We've got data!
    classic.setHighRes(!hr);
    hr = !hr;
    delay(1000);
		classic.printDebug();  // Print all of the values!
	}
```

Edited to this snipped instead, as it is less crappy :P

```cpp
bool hr = false;

void loop() {
  boolean success = classic.update();  // Get new data from the controller

  if (success == true) {  // We've got data!
    classic.setHighRes(hr);
    classic.update();
    Serial.println("---");
    Serial.println(hr ? "High Res:" : "Low Res:");
    classic.printDebug();  // Print all of the values!
    Serial.println("---");
    hr = !hr;
    delay(1000);
  }
```

Current `flex` output:

(While holding A and Left Dpad)

```
---
High Res:
Classic ____ | ___ | A___ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)
---
---
Low Res:
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:239_ Z:LR | (HR)
---
---
High Res:
Classic ____ | ___ | A___ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)
---
---
Low Res:
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:239_ Z:LR | (HR)
---
---
High Res:
Classic ____ | ___ | A___ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)
---
```

Current Master:

```
---
Low Res:
Classic <^v> | -H+ | ABXY L:( 28,   0) R:(  5,   0) | LT:  4X RT:  0X Z:LR
---
---
High Res:
Classic ___> | _H+ | __XY L:(135,  16) R:( 32,   0) | LT:255X RT:239_ Z:LR (High Res)
---
---
Low Res:
Classic <^v> | -H+ | ABXY L:( 28,   0) R:(  5,   0) | LT:  4X RT:  0X Z:LR
---
---
High Res:
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:239_ Z:LR (High Res)
---
```

Though, now it seems to just be crapping the bed. So maybe just ignore me, haha.

dmadison, 2020-06-07:
>Classic ____ | ___ | ____ L:(132, 134) R:(134, 134) | LT: 0_ RT: 0_ Z:__ | (HR)

Thanks. That's within the margin, so I'm going to call that 'good'.

---

Re: the "swapping" output:

>Classic ____ | ___ | A___ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)

The `| (HR)` bit at the end in the `flex` branch means that the internal "high res" flag is set and it's using the high resolution maps, so the `check` function is doing its job. I'd check with a logic analyzer, but my guess is that the controller is just returning junk.

Does it work using the basic `debugPrint` example without the `setHighRes` stuff in the loop? My hunch is that it doesn't like the extra register writes.

nullstalgia, 2020-06-07:
Oh, yeah

It works fine, minus the 

> †† - DPad acts like L stick, but that's just the adapter. 

And I was wondering if it did the same when HighRes was turned off, but didn't have any luck testing.

dmadison, 2020-06-07:
Just to make sure I'm understand: the controller works fine, normal output and everything, but starts returning garbage if you attempt to set the 'high res' mode a second time?

nullstalgia, 2020-06-07:
https://gist.github.com/nullstalgia/458d5a691ef6ab88482307b3b45377e0

Sketch I'm using on both `flex` and `master` branches.

---

With Flex:

Genuine Classic:

```
SWAPPING TO: High Res
Classic ____ | ___ | ____ L:(124, 120) R:(125, 132) | LT: 24_ RT: 22_ Z:__ | (HR)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(125, 132) | LT: 24_ RT: 22_ Z:__ | (HR)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(125, 132) | LT: 24_ RT: 22_ Z:__ | (HR)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(125, 132) | LT: 24_ RT: 22_ Z:__ | (HR)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(125, 132) | LT: 24_ RT: 22_ Z:__ | (HR)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(125, 132) | LT: 24_ RT: 22_ Z:__ | (HR)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(125, 132) | LT: 24_ RT: 22_ Z:__ | (HR)
...
trimmed
...
SWAPPING TO: Low Res
Classic ____ | ___ | ____ L:(124, 120) R:(120, 128) | LT: 24_ RT: 16_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(120, 128) | LT: 24_ RT: 16_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(120, 128) | LT: 24_ RT: 16_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(120, 128) | LT: 24_ RT: 16_ Z:__
```

Hyperkin Adapter (worked well on updates before this hi res stuff):

```
SWAPPING TO: High Res
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ | (HR)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
...
trimmed
...
SWAPPING TO: Low Res
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:255_ Z:LR | (HR)
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:255_ Z:LR | (HR)
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
```

---

Master branch:

Genuine Classic:

```
SWAPPING TO: High Res
Classic ____ | ___ | ____ L:( 62, 205) R:(125, 132) | LT: 26_ RT: 22_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:( 25, 123) R:(125, 132) | LT: 26_ RT: 22_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:( 79,  40) R:(125, 132) | LT: 26_ RT: 22_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 121) R:(125, 132) | LT: 26_ RT: 22_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 121) R:(125, 132) | LT: 26_ RT: 22_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 121) R:(125, 132) | LT: 26_ RT: 22_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 121) R:(125, 132) | LT: 26_ RT: 22_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(125, 132) | LT: 26_ RT: 22_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(124, 120) R:(125, 132) | LT: 26_ RT: 22_ Z:__ (High Res)
...
trimmed
...
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
SWAPPING TO: Low Res
Classic ____ | ___ | ____ L:( 31,  30) R:( 15,  16) | LT:  3_ RT:  2_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ____ | ___ | ____ L:( 31,  30) R:( 15,  16) | LT:  3_ RT:  2_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ____ | ___ | ____ L:( 31,  30) R:( 15,  16) | LT:  3_ RT:  2_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
```

Hyperkin Adapter:

```
SWAPPING TO: High Res
Classic ___> | _H+ | __XY L:(160,  16) R:( 32,   0) | LT:255X RT:255_ Z:LR (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ (High Res)
ID: 0x00 0x00 0xA4 0x20 0x03 0x01
Classic ____ | ___ | ____ L:(128, 128) R:(128, 128) | LT:  0_ RT:  0_ Z:__ (High Res)
...
trimmed
...
SWAPPING TO: Low Res
Classic <^v> | -H+ | ABXY L:(  0,   0) R:( 21,   0) | LT:  4X RT:  0X Z:LR
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ____ | ___ | ____ L:( 32,  32) R:( 16,  16) | LT:  0_ RT:  0_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ____ | ___ | ____ L:( 32,  32) R:( 16,  16) | LT:  0_ RT:  0_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ____ | ___ | ____ L:( 32,  32) R:( 16,  16) | LT:  0_ RT:  0_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
Classic ____ | ___ | ____ L:( 32,  32) R:( 16,  16) | LT:  0_ RT:  0_ Z:__
ID: 0x00 0x00 0xA4 0x20 0x01 0x01
```

dmadison, 2020-06-07:
>The spikes at the beginning of the mode swap is probably just the result of a missing update.

Either that or it's taking its sweet time to implement the changed mode.

Because the "standard" data looks different on the flex branch vs. the master I have a sneaking suspicion that it doesn't like us requesting 8 bytes of data while it's in the "standard" control mode. Easy enough to check on the `master` branch: after setting the mode back to standard, change the request size (`setRequestSize`) to 8 and see what it does. But if that's the case I don't think there's anything we can do about it.

nullstalgia, 2020-06-07:
>  change the request size (`setRequestSize`) to 8 and see what it does

Seems to work fine when switching request size to 8

dmadison, 2020-06-07:
After setting it to 'high res' and then back to 'standard'?

I don't know what to tell you then - those two 8 byte requests are the only difference the controller sees between the two branches.

nullstalgia, 2020-06-07:
Found the problem:

Last 2 bytes of the 8-byte request is not `0x00 0x00`, thus the switch of mode occurs, but does not change `highRes` as `currentMode` in `setDataMode` still says "Yes, he's in high res!" due to that.

Which brings me again to my `force` suggestion (which isn't really needed for normal usage, but for stuff like this, testing controllers, it could be useful and adds no overhead/extra code if its never used)

Before change:

```
Sketch uses 10922 bytes (38%) of program storage space. Maximum is 28672 bytes.
Global variables use 696 bytes (27%) of dynamic memory, leaving 1864 bytes for local variables. Maximum is 2560 bytes.
```

After:

```
Sketch uses 10922 bytes (38%) of program storage space. Maximum is 28672 bytes.
Global variables use 696 bytes (27%) of dynamic memory, leaving 1864 bytes for local variables. Maximum is 2560 bytes.
```

And when being used:

```
Sketch uses 10928 bytes (38%) of program storage space. Maximum is 28672 bytes.
Global variables use 696 bytes (27%) of dynamic memory, leaving 1864 bytes for local variables. Maximum is 2560 bytes.
```

Suggested change:

`ClassicController.h`

```cpp
boolean setHighRes(boolean hr = true, boolean force = false);
```

`ClassicController.cpp`

```cpp
boolean ClassicController_Shared::setHighRes(boolean hr, boolean force) {
	// 'success' if the mode is changed to the one we're trying to set
    if (force == false){
        return setDataMode(hr) && (getHighRes() == hr);
    } else {
        setDataMode(hr);
        highRes = hr;
        return true;
    }
}
```

nullstalgia, 2020-06-07:
Well actually, it would be better if it was more like

```cpp
        if(setDataMode(hr)){
            highRes = hr;
            return true;
        }
        return false;
```

To account for `setDataMode` not working, but you get my point. :)

dmadison, 2020-06-07:
>Found the problem:
>Last 2 bytes of the 8-byte request is not 0x00 0x00, thus the switch of mode occurs, but does not change highRes as currentMode in setDataMode still says "Yes, he's in high res!" due to that.

Ah I see. You were manipulating the control surfaces during the test and that was throwing me off; I thought the controller was reporting bad data.

Unfortunately I don't think there's anything to be done about that `0x00` issue. The other option is to use the data maps and check if the control surfaces are within a specific range, but that's more intensive and much less reliable.

>Which brings me again to my force suggestion (which isn't really needed for normal usage, but for stuff like this, testing controllers, it could be useful and adds no overhead/extra code if its never used)

Okay, I've added that in. The logic for the argument is inverted because it's disabling the verification check rather than 'forcing' the mode - although the effect is the same.

---

It looks as though the branch is a success though and solves the issue: the `setHighRes` function now only returns 'true' if the requested mode is set (as well as can be determined). Moreover, for all of the controllers tested the class works out of the box without any extra user input and all of the data ranges are now consistent regardless of mode. I'd call that a smashing success.

I'll try to set aside some time tomorrow to create the PR and push a new release 🥳.

nullstalgia, 2020-06-07:
The only controller I wish I could have tested is those really *really* crappy wireless ones that I ended up returning, but they should work. (Given they had the 0x03 ID, and seemed to follow the high res data format)

---

Anyway, I ordered an (admittedly, third party) Taiko no Tatsujin Controller (often shortened to Tatacon) for the Wii. It will be based off of this adapter: https://github.com/mon/TataconUSB and my own findings/testing to make it compatible with this library and it's flow.

PR #64 (2020-06-08): Classic Controller Flex
This pull request changes the behavior of the `ClassicController` class to automatically switch ("flex") between the high resolution and standard mappings depending on what the controller supports. The goal is to allow *all* "Classic Controller" type controllers to connect and report valid data without requiring the user to intervene to set the data mode.

* Classic Controllers now attempt to set "high res" mode on connect and will revert to standard mappings if the request isn't successful.
* After setting the data mode, the control data is read back and evaluated to c [... cut]
closed 2020-06-08 by dmadison