dmadison/NintendoExtensionCtrl #68: Non genuine classic controller not initialised with classic.begin(); but can get data with controller.begin();

joshhh, 2021-03-07:
Trying to find out what's wrong with my XBOX controller emulation on chinese arduino micro 32u4 board mimicking as leonardo board

Always get Controller not detected with classic.begin(); but can get data with controller.begin();

All RAW data from controller detected in serial monitor, so it is working. 

Also everything is working with Leonardo WiiClassicController GamePad by Nguyen Chi Tam

dmadison, 2021-03-07:
A few questions:
* Which examples are working? Which examples are not?
* What controller are you using (make and model)?
* Are you using a breakout board? If so, which?
* What are your hardware connections to the controller?
* With `classic.begin()` returning false, are you still able to get data in subsequent requests?

joshhh, 2021-03-07:
DebugPrint.ino is working 100% with all raw data visible and reacting on button-pads 
Classic_DebugPrint.ino get Classic Controller not detected all the time

Arduino board https://www.aliexpress.com/item/32890044536.html
Classic controller clone  https://www.aliexpress.com/item/1005001430909586.html

No breakout boards are used.  Only diy pcb as connector to wii controller and direct soldering to SDA, SCL, GND, VCC

With classic.begin() I get Classic Controller not detected
With controller.begin(); I get raw data...

dmadison, 2021-03-08:
Hmm. When used as a `ClassicController` object the code jumps through a few hoops to get both Classic Controllers and NES / SNES Mini controllers to behave the same. If it's falling the 'connect' check as a `ClassicController` but working as a generic one then that controller-specific initialization must be failing for some reason, which it should only do if the I²C functions report a problem. Your clone controller is probably not behaving like the genuine ones and doing something it's not supposed to.

If you run the Classic Controller example but ignore the result of the `connect()` call it should work, but that's not a great solution.

Try adding this block to the `DebugPrint` example, after the `connect()` call:

```
  if(controller.writeRegister(0xFE, 0x03)) {
     Serial.println("Successful register write");
  }
  else {
    Serial.println("Failed register write");
  }
```

Run that and let me know the output.

Then replace it with this, in the same place after the `connect()` call, and run it as well:
```
  uint8_t temp[8];
  if(controller.requestData(0x00, 8, temp)) {
     Serial.println("Successful read");
  }
  else {
     Serial.println("Failed read");
  }
```

One of the two, possibly both, is probably failing.

joshhh, 2021-03-08:
First code inside void loop() sequence receive this

```
15:09:09.467 -> Failed register write
15:09:09.467 -> Raw[6]: 0xA0 0x20 0x10 0x00 0xFF 0xFF
```

Second 

```
15:09:21.599 -> Successful read
15:09:21.599 -> Raw[6]: 0xA0 0x20 0x10 0x00 0xFF 0xFF
```

So first one cause it is fail to connect

dmadison, 2021-03-08:
Okay great, thanks. I'll have to do some thinking about how best to implement a workaround for that.

If you could, in the same place try these two lines:

    controller.writeRegister(0xFE, 0x03);
    controller.setRequestSize(8);

And please post the output from the first 'Raw' debug message.

joshhh, 2021-03-08:
Got this. last two bytes are zeroes 

`#include <NintendoExtensionCtrl.h>

ExtensionPort controller;

void setup() {
	Serial.begin(115200);
	controller.begin();

	while (!controller.connect()) {
		Serial.println("Controller not detected!");
		delay(1000);
	}
}

void loop() {
	boolean success = controller.update();  // Get new data from the controller

	controller.writeRegister(0xFE, 0x03);
	controller.setRequestSize(8);
	delay(100);

	if (success == true) {  // We've got data!
		controller.printDebugRaw();  // Print all of the values!
	}
	else {  // Data is bad :(
		Serial.println("Controller Disconnected!");
		delay(1000);
		controller.connect();
	}
}`

`18:04:58.595 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:58.707 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:58.819 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:58.925 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:58.992 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:59.098 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:59.208 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:59.320 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:59.428 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:59.539 -> Raw[8]: 0x80 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:59.611 -> Raw[8]: 0x80 0x0C 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:59.717 -> Raw[8]: 0x80 0x00 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:59.824 -> Raw[8]: 0xA0 0x00 0x10 0x00 0xFF 0xFF 0x00 0x00
18:04:59.930 -> Raw[8]: 0xA0 0x00 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.036 -> Raw[8]: 0xBF 0x00 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.145 -> Raw[8]: 0xBF 0x00 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.250 -> Raw[8]: 0xBF 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.357 -> Raw[8]: 0xA2 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.461 -> Raw[8]: 0xA0 0x38 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.529 -> Raw[8]: 0x81 0x3D 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.637 -> Raw[8]: 0x80 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.743 -> Raw[8]: 0x80 0x0F 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.848 -> Raw[8]: 0x80 0x00 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:00.954 -> Raw[8]: 0x60 0x80 0x17 0x00 0xFF 0xFF 0x00 0x00
18:05:01.065 -> Raw[8]: 0x3F 0x00 0x1F 0x00 0xFF 0xFF 0x00 0x00
18:05:01.170 -> Raw[8]: 0x3F 0x00 0x1F 0x00 0xFF 0xFF 0x00 0x00
18:05:01.275 -> Raw[8]: 0x3F 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:01.381 -> Raw[8]: 0x7F 0x20 0x80 0x00 0xFF 0xFF 0x00 0x00
18:05:01.489 -> Raw[8]: 0xFB 0xF3 0x80 0x00 0xFF 0xFF 0x00 0x00
18:05:01.560 -> Raw[8]: 0xE0 0xF8 0x90 0x00 0xFF 0xFF 0x00 0x00
18:05:01.667 -> Raw[8]: 0xC1 0xEF 0x9F 0x00 0xFF 0xFF 0x00 0x00
18:05:01.774 -> Raw[8]: 0x80 0x55 0x1F 0x00 0xFF 0xFF 0x00 0x00
18:05:01.880 -> Raw[8]: 0xA0 0x00 0x1F 0x00 0xFF 0xFF 0x00 0x00
18:05:01.987 -> Raw[8]: 0x3B 0x00 0x1D 0x00 0xFF 0xFF 0x00 0x00
18:05:02.089 -> Raw[8]: 0x3F 0x00 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:02.195 -> Raw[8]: 0x3F 0x20 0x00 0x00 0xFF 0xFF 0x00 0x00
18:05:02.302 -> Raw[8]: 0xF0 0xE8 0x85 0x00 0xFF 0xFF 0x00 0x00
[... 75 lines cut ...]
18:05:10.105 -> Raw[8]: 0xA0 0x20 0x10 0x1F 0xF9 0xFF 0x00 0x00
18:05:10.210 -> Raw[8]: 0xA0 0x20 0x10 0x1F 0xF1 0xFB 0x00 0x00
18:05:10.318 -> Raw[8]: 0xA0 0x20 0x70 0xE0 0xDF 0xFB 0x00 0x00
18:05:10.392 -> Raw[8]: 0x89 0x04 0x72 0xE0 0xDF 0xFF 0x00 0x00
18:05:10.498 -> Raw[8]: 0xC0 0xC0 0xF6 0xE0 0xDF 0xFF 0x00 0x00
18:05:10.604 -> Raw[8]: 0xD1 0xCA 0xF0 0xE0 0xDF 0xFF 0x00 0x00
18:05:10.712 -> Raw[8]: 0xA0 0x3F 0x70 0xE0 0xDF 0xFF 0x00 0x00
18:05:10.817 -> Raw[8]: 0xA0 0x3F 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:10.925 -> Raw[8]: 0xA0 0x3F 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.026 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.134 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.246 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.314 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.427 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.536 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.642 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.752 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.857 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:11.963 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:12.036 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:12.141 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:12.253 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:12.358 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
18:05:12.468 -> Raw[8]: 0xA0 0x20 0x10 0x00 0xFF 0xFF 0x00 0x00
`

dmadison, 2021-03-08:
Great, that's exactly what it should be reporting. That means that while it NACK's the request to set the data mode because it doesn't understand it, it still reports its data in a way that we can figure out what mode it's in.

I've added a patch to the [`classic-clone-fix` branch](../tree/classic-clone-fix). Please download and test that to see if it fixes the issue.

joshhh, 2021-03-08:
Confirmed. It is working now and output to serial monitor all pressed buttons events. Also XInput XBOX controller code also works now in control panel and other applications

dmadison, 2021-03-08:
Fantastic. Thanks for reporting the issue and I'm glad I was able to fix it for you. I'll merge those changes and push out a new release of the library.

Have fun with the controller!

PR #69 (2021-03-08): Fix for Classic Controller clones without data mode support
It appears that some Classic Controller clones designed before the "high resolution" mode was discovered do not properly acknowledge requests sent to the data mode register (`0xFE`). As a result of this those controllers NACK the data mode register write, causing the `ClassicController` `connect()` function to return an error and inform the user that the controller is not connected. Normally the request would only be NACK'd if the controller isn't connected to the bus, so this is seen as a communication error.

This change adds a seco [... cut]
closed 2021-03-08 by dmadison