dmadison/NintendoExtensionCtrl #72: I2C pins a beginner's question

nonaak, 2022-04-11:
Thank you for all the work you have done to create this library.

I would like to get the nunchuk working on my esp-01. only then does the ESP not see the nunchuk. are the I2C pins fixed in the standard I2C pins of an ESP in your code? Can I change it or do I have to do it another way?

Now I have it working on a "regular" ESP8266 with indicated I2C pins. I can use it to control my robot via the UDP.
But I like to keep it as small as possible, to be able to put everything in the nunchuk (Li-Po + ESP)

Thanks for reading my post, it will be totally awesome if you push me in the right direction a bit

dmadison, 2022-04-11:
Hi there. Each controller object takes a reference to the built-in `Wire` class which handles the lower level I²C commands. From what I remember the ESP uses software I²C, and you should be able to change the assigned pins through a function of the `Wire` class. Take a look at the ESP core for Arduino, the function you need should be in there.

nonaak, 2022-04-13:
I find it out. in "Void setup" put "Wire.pins(0, 2);" and it works. add olso 10k resistors between 3.3v and gpo0 and gpo2 for I2C pullup.

Maby as sombody have the same qwestion

dmadison, 2022-04-13:
Glad you were able to figure it out!

For what it's worth, the controllers themselves have strong pull-up resistors built-in (usually 4.7k or stronger). If the microcontroller doesn't have pull-ups on its own bus, a weak pull-up is recommended. I've used 47k in the past with a lot of success.

closed 2022-04-13 by dmadison